gaowenliang/imu_utils #2: program crash

pxlong, 2018-03-12:
I run this package with my imu data, and got this error: 

Gyro X 
C   -1.15909    18.9571   -3.72175    3.77855 -0.0240066
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0310 18:00:55.632755 28994 solver.cc:487] Terminating: Invalid configuration. Solver::Options::max_lbfgs_rank = 0. Violated constraint: Solver::Options::max_lbfgs_rank > 0
 Bias Instability 6.07778e-05 rad/s, at 5.04982 s
 White Noise 9.55941e-05 rad/s
[imu_an-1] process has died [pid 28994, exit code -11, cmd /home/pxlong/ws/catkin_ws/devel/lib/imu_utils/imu_an __name:=imu_an __log:=/home/pxlong/.ros/log/fbb220fa-1c4e-11e8-86a2-7085c22aa66f/imu_an-1.log].
log file: /home/pxlong/.ros/log/fbb220fa-1c4e-11e8-86a2-7085c22aa66f/imu_an-1*.log

gaowenliang, 2018-03-12:
my environment set: Eigen 3.3.4 + Ceres 1.13.0 + Ubuntu 16.04 + ROS Kinetic, a little bit different with yours. 
Seems there are something wrong with the Hessian Matrix in your solver. 
would you mind providing your rosbag data to me?

pxlong, 2018-03-13:
@gaowenliang  I have tested the repo on my computer with the data (16448) you provided. 
But I still got this error.

I will try to update Ceres to 1.13.0, and let you know the results.

pxlong, 2018-03-13:
@gaowenliang After updating Ceres to 1.13.0, everything is OK.
Thanks.

closed 2018-03-13 by pxlong