gaowenliang/imu_utils #3: The result error

fanhanyun, 2018-05-09:
Thanks a lot ！！！
I record a rosbag and play it，but the Gyro z of the result was error，and I dont know how to solve that.
here is a part of my result.(ps: Please forgive my bad chinglish :) )
==========================
Gyro X 
C -28945.1   194378  7.23339  139.054 -3.53494
 Bias Instability 0.00484778 rad/s, at 0.104268 s
 White Noise 2.94973 rad/s
  bias -1.88767 degree/s
-------------------
Gyro y 
C -3460.61  44971.8  457.603   -18.87 0.130777
 Bias Instability 0.00373774 rad/s, at 3416.76 s
 White Noise 0.676594 rad/s
  bias 1.82403 degree/s
-------------------
Gyro z 
C 0 0 0 0 0
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0509 10:56:09.249119 25327 residual_block.cc:131] 

Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all    
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation. 

Residual Block size: 1 parameter blocks x 1 residuals

For each parameter block, the value of the parameters are printed in the first column   
and the value of the jacobian under the corresponding residual. If a ParameterBlock was 
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry 
of the Jacobian/residual array was requested but was not written to by user code, it is 
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating 
to Inf or NaN is also an error.  

Residuals:             -nan 

Parameter Block 0, size: 5

           0 |         -nan 
           0 |         -nan 
           0 |         -nan 
           0 |         -nan 
           0 |         -nan 

E0509 10:56:09.249228 25327 trust_region_minimizer.cc:72] Terminating: Residual and Jacobian evaluation failed.
 Bias Instability 0 rad/s, at 3416.76 s
 White Noise 0 rad/s
  bias 0 degree/s

gaowenliang, 2018-05-10:
seems that data you collected is not suitable for the Ceres Solver. Can you provide your data for debug?

fanhanyun, 2018-05-11:
alright...this error just because of my c++program，it didn't read the last data。
anyway,thank you for you reply

closed 2018-05-11 by gaowenliang

Lmixq, 2021-01-19:
Hi, I meet the same problem. Do you solve it? Thank you!

Diggle221, 2022-01-11:
> seems that data you collected is not suitable for the Ceres Solver. Can you provide your data for debug?

Hello, I also encountered the above problems. This is my bag. How can I solve them? Thank you very much.
[https://drive.google.com/file/d/1if5WapgQ524ZZpxdT1di73sIZxgrIq_n/view?usp=sharing](url)