gaowenliang/imu_utils #6: Idea to improve the installation step in the README.md

ferreram, 2018-07-13:
Hi,
Thanks a lot for your library!

In order to help people use it easily, I think you could add the following indications in the installation step:
- Required: apt-get install libdw-dev
- In code_utils: in mat_io_test.cpp and sumpixel_test.cpp => change ' #include "backward.hpp" ' into ' #include "code_utils/backward.hpp" '

Cheers

Yannnnnnnnnnnn, 2018-08-07:
Meet the same problems.

fathyRajab, 2018-08-26:
@ferreram Thanks a lot! That helped.

mzahana, 2018-08-26:
@ferreram Thanks! That helped!

gaowenliang, 2018-08-27:
fixed, thank you

closed 2018-08-27 by gaowenliang

mzahana, 2018-09-03:
@gaowenliang I had some more issues during installation, and here how is I fixed it.

[Ceres](http://ceres-solver.org/installation.html#linux) solver is required.
Steps to install Ceres:
```bash
sudo apt-get install libgoogle-glog-dev
sudo apt-get install libatlas-base-dev
sudo apt-get install libeigen3-dev
```
Download the following file:
```
http://ceres-solver.org/ceres-solver-1.14.0.tar.gz
```

Execute:
```bash
tar zxf ceres-solver-1.14.0.tar.gz
mkdir ceres-bin
cd ceres-bin
cmake ../ceres-solver-1.14.0
make -j3
make install
```

Also, in [sumpixel_test.cpp](https://github.com/gaowenliang/code_utils/blob/master/src/sumpixel_test.cpp#L2), I had to change 
```
#include "backward.hpp"
```
to
```
#include "code_utils/backward.hpp"
```

Thanks.