gaowenliang/imu_utils #11: Question on the calibration results on your given data.

23matin, 2018-11-05:
Hi Wenliang Gao,
I run your code with the the default launch file and the bagfiles on the Baidu Netdisk,but I got different results with your results. Especially the gyr_n and acc_n. Could you please help me explain why? Thanks!!
Here are some results I got : 
```
type: IMU
name: gx4
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 1.3776557135598267e-03
      gyr_w: 4.4980238465211471e-05
   x-axis:
      gyr_n: 1.3598916608690695e-03
      gyr_w: 4.3237605124366711e-05
   y-axis:
      gyr_n: 1.4085032617136594e-03
      gyr_w: 4.7203631246970414e-05
   z-axis:
      gyr_n: 1.3645722180967506e-03
      gyr_w: 4.4499479024297288e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 1.9804078615439449e-02
      acc_w: 4.2787561680514757e-04
   x-axis:
      acc_n: 2.1200169501503625e-02
      acc_w: 3.7692313329243398e-04
   y-axis:
      acc_n: 1.9195904275601066e-02
      acc_w: 4.8504332057982245e-04
   z-axis:
      acc_n: 1.9016162069213648e-02
      acc_w: 4.2166039654318628e-04
```
```
type: IMU
name: xsens
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 1.5766795105350902e-03
      gyr_w: 5.8186310639220816e-05
   x-axis:
      gyr_n: 1.5913990126303940e-03
      gyr_w: 6.1777289031013914e-05
   y-axis:
      gyr_n: 1.5285453823580789e-03
      gyr_w: 6.7252832784928527e-05
   z-axis:
      gyr_n: 1.6100941366167980e-03
      gyr_w: 4.5528810101720015e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 7.7486924587566184e-03
      acc_w: 3.0142621178774011e-04
   x-axis:
      acc_n: 8.0543421537212501e-03
      acc_w: 4.4048166959126769e-04
   y-axis:
      acc_n: 7.6276107736174539e-03
      acc_w: 1.9648598133533297e-04
   z-axis:
      acc_n: 7.5641244489311511e-03
      acc_w: 2.6731098443661977e-04
```
```
type: IMU
name: N3
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 2.0474063562222508e-03
      gyr_w: 4.4107187176933610e-05
   x-axis:
      gyr_n: 1.9934577058496185e-03
      gyr_w: 4.3455951372090669e-05
   y-axis:
      gyr_n: 2.0623111914684425e-03
      gyr_w: 5.1949820903083133e-05
   z-axis:
      gyr_n: 2.0864501713486914e-03
      gyr_w: 3.6915789255627028e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 2.7361865573858338e-02
      acc_w: 6.9869639650317790e-04
   x-axis:
      acc_n: 2.2915909292582394e-02
      acc_w: 6.2346942228045700e-04
   y-axis:
      acc_n: 2.1552485668793139e-02
      acc_w: 5.4394057410119575e-04
   z-axis:
      acc_n: 3.7617201760199490e-02
      acc_w: 9.2867919312788095e-04
```
```
type: IMU
name: "16448"
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 2.5556549916784113e-03
      gyr_w: 7.4036973730112702e-05
   x-axis:
      gyr_n: 2.5752367201719532e-03
      gyr_w: 7.9965694430990478e-05
   y-axis:
      gyr_n: 2.5521145069793234e-03
      gyr_w: 6.8137921217407534e-05
   z-axis:
      gyr_n: 2.5396137478839575e-03
      gyr_w: 7.4007305541940135e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 2.9005443699517627e-02
      acc_w: 1.2919311021553450e-03
   x-axis:
      acc_n: 2.6832688337940648e-02
      acc_w: 9.0581978608079256e-04
   y-axis:
      acc_n: 3.4875656476624710e-02
      acc_w: 2.0564271053782313e-03
   z-axis:
      acc_n: 2.5307986283987522e-02
      acc_w: 9.1354641500701139e-04
```

Peach-ran, 2019-05-11:
i have the same question with you, can anybody help?

gaowenliang, 2019-06-21:
Uhhhh，seems something different
I will check it later

robotdream, 2019-06-21:
> Uhhhh，seems something different
> I will check it later

I also observed some difference between my results and the results posted on Github. 
For A3 data, I always got the following results no matter how fast I played the rosbag.

%YAML:1.0
---
type: IMU
name: A3
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 2.1351484255583693e-03
      gyr_w: 3.0118327235660890e-05
   x-axis:
      gyr_n: 2.2761455758382055e-03
      gyr_w: 3.5807139355605253e-05
   y-axis:
      gyr_n: 2.1481099566996453e-03
      gyr_w: 3.1987662852364106e-05
   z-axis:
      gyr_n: 1.9811897441372574e-03
      gyr_w: 2.2560179499013303e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 2.7433396928041465e-02
      acc_w: 6.0511514459808539e-04
   x-axis:
      acc_n: 2.2944432766196329e-02
      acc_w: 5.3089385840146867e-04
   y-axis:
      acc_n: 2.4074942631450987e-02
      acc_w: 6.0364353027662736e-04
   z-axis:
      acc_n: 3.5280815386477080e-02
      acc_w: 6.8080804511616024e-04

tanshiheng, 2019-10-17:
I also observed samedifference between gyr_n and acc_n，about 10x：
my result：
%YAML:1.0
---
type: IMU
name: A3
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 2.1368945822378715e-03
      gyr_w: 3.0200858616831391e-05
   x-axis:
      gyr_n: 2.2850607115752581e-03
      gyr_w: 3.5888806047445071e-05
   y-axis:
      gyr_n: 2.1437755224791373e-03
      gyr_w: 3.2187208542462374e-05
   z-axis:
      gyr_n: 1.9818475126592186e-03
      gyr_w: 2.2526561260586737e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 2.7615339617476531e-02
      acc_w: 6.3638852544213207e-04
   x-axis:
      acc_n: 2.3509501859500022e-02
      acc_w: 5.3422099556560550e-04
   y-axis:
      acc_n: 2.3564766827602595e-02
      acc_w: 6.0043815190390657e-04
   z-axis:
      acc_n: 3.5771750165326979e-02
      acc_w: 7.7450642885688447e-04

chenyumingx, 2019-11-21:
My result is also different from that given.Here's what I got

%YAML:1.0
---
type: IMU
name: "A3"
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 2.1267737833765769e-03
      gyr_w: 3.0020942712407391e-05
   x-axis:
      gyr_n: 2.2545903384908663e-03
      gyr_w: 3.4578244608146821e-05
   y-axis:
      gyr_n: 2.1446706374840321e-03
      gyr_w: 3.2559043638179981e-05
   z-axis:
      gyr_n: 1.9810603741548327e-03
      gyr_w: 2.2925539890895362e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 2.7277780651013772e-02
      acc_w: 6.4735219294483461e-04
   x-axis:
      acc_n: 2.2893552216220955e-02
      acc_w: 5.2617622561120197e-04
   y-axis:
      acc_n: 2.2754690562618862e-02
      acc_w: 6.0703818233019756e-04
   z-axis:
      acc_n: 3.6185099174201506e-02
      acc_w: 8.0884217089310429e-04

geerniya, 2020-11-18:
I have the same result. But I think the reason is that the IMU is not flat.
---
header: 
  seq: 245185
  stamp: 
    secs: 1518107778
    nsecs: 380825295
  frame_id: "FLU"
orientation: 
  x: -0.0107081532954
  y: -0.588978761633
  z: -0.000865663655542
  w: 0.808077142277
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00176756794099
  y: -0.00166124606039
  z: 0.000752829015255
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 9.2817324084
  y: -0.159711141288
  z: 2.98692894071
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---