gaowenliang/imu_utils #12: Failed to build the package

LeungHEX, 2019-01-11:
Once  follow the constrction in the readme, but the bulid console show the error msg like :+1: 

- Could not find a package configuration file provided by "code_utils" with
- any of the following names:
- 
- code_utilsConfig.cmake
- code_utils-config.cmake

then i try to build the code_utils without imu_utils it shows the following error:

- fatal error: backward.hpp: No such file or directory

I am pretty sure that the files pull from your git is correct and complete, but why it cannot build?

Or can someone else help me?

LeungHEX, 2019-01-11:
Besides, I copy the backward.hpp to where the cpp need, it built sucessfully. But I have to build code_utils first then add the imu_utils into the workspace, it seems that this process is not the same as you introduced in the readme, is there something wrong in my env? Thanks in advance if there is any useful tips.

Young532042725, 2019-03-18:
@LeungHEX , hi, my friend,  i also meet the problem as yours, have you solved it? would you please give me some help? thank U.

Sayheyheyhey, 2019-03-18:
像上一条评论里写的一样
只需要把sumpixel_test.cpp文件中的下边这句
#include "backward.hpp"
改成这句
#include "code_utils/backward.hpp"
就解决了

Young532042725, 2019-03-18:
@Sayheyheyhey ok 了 谢谢您

luzhaozhang2276, 2019-06-22:
@Sayheyheyhey  Using your method, the problem is still unresolved.
I compile code_utils first, then add imu_utils in, compile again, and it succeeds.

beduffy, 2019-08-12:
Yep, compiling with code_utils only in the src folder worked (after fixing some dependencies e.g. backward.hpp, libf77blas.so, libspqr.so) and then I moved imu_utils to the same folder and ran catkin_make again. This should be documented or fixed for other people.

Hezip, 2020-03-07:
This is a problem with the **compile chain**. 
You need to add such a sentence in the **package.yaml** of **imu_utils** 
**<buildtool_depend> code_utils </ buildtool_depend>**

Tianci-Wen, 2023-02-02:
> Yep, compiling with code_utils only in the src folder worked (after fixing some dependencies e.g. backward.hpp, libf77blas.so, libspqr.so) and then I moved imu_utils to the same folder and ran catkin_make again. This should be documented or fixed for other people.

I solved the problems by this way.