gaowenliang/imu_utils #13: can't locate node [imu_an]

xuAn1604, 2019-03-11:
Hello Mr.Gao and everyone,
    I had downloaded this code and build it in my catkin workspace. My operating system is Ubuntu16.04. When I try to launch A3.launch, I got this error:
    `ERROR: cannot launch node of type [imu_utils/imu_an]: can't locate node [imu_an] in package [imu_utils]`
    The build information is:
    `Base path: /home/zxchang/imu_ws
Source space: /home/zxchang/imu_ws/src
Build space: /home/zxchang/imu_ws/build
Devel space: /home/zxchang/imu_ws/devel
Install space: /home/zxchang/imu_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/zxchang/imu_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/zxchang/imu_ws/build"
####
[ 15%] Built target polynomial
[ 46%] Built target cv_utils
[ 46%] Built target sumpixel_test
[ 61%] Built target matIO_test
[100%] Built target pnp
`
Appreciate anyone's help.

andre-nguyen, 2019-03-27:
Did you source your workspace...

allenhsu6, 2019-06-18:
@xuAn1604 hi, have you solved this problem? i have met the same one.

allenhsu6, 2019-06-18:
@xuAn1604   try to catkin_make the package:imu_utils alone.
in your workspace: 
`catkin_make  -DCATKIN_WHITELIST_PACKAGES="imu_utils" `

it's works for me

AfterJourney00, 2020-08-12:
You have to source your worksapce like:

cd [your workspace]
source devel/setup.bash

houcemsaidani, 2022-04-27:
> 
it does not work, here's the error:

Could not find a package configuration file provided by "code_utils" with
  any of the following names:

    code_utilsConfig.cmake
    code_utils-config.cmake

  Add the installation prefix of "code_utils" to CMAKE_PREFIX_PATH or set
  "code_utils_DIR" to a directory containing one of the above files.  If
  "code_utils" provides a separate development package or SDK, be sure it has
  been installed.

ZhangzrJerry, 2024-11-15:
> @xuAn1604 try to catkin_make the package:imu_utils alone. in your workspace: `catkin_make -DCATKIN_WHITELIST_PACKAGES="imu_utils" `
> 
> it's works for me

same, I execute commands below, and it works 
```
catkin_make -DCATKIN_WHITELIST_PACKAGES="imu_utils"
catkin_make -DCATKIN_WHITELIST_PACKAGES="code_utils" 
```