gaowenliang/imu_utils #15: Wait for imu data

csl1994, 2019-04-18:
Hello Gao，
May I ask you a question？After run launch file，“wait for imu data” will appear. I check topic information(rostopic list  & rostopic inof /xx), there is not any problem. 
Thank you very much！

johnjin404, 2019-04-20:
you need to wait for the bad file playing end,,look at the imu_an.cpp.

csl1994, 2019-04-21:
> you need to wait for the bad file playing end,,look at the imu_an.cpp.
thanks!

maxwu8416, 2019-05-06:
@csl1994 hay~ i also have same problem as you did. do you solved the problem ? @johnjin404  i already your suggest  try to wait for bag file playing until the end but nothing change. is it need to do it at the same time ? (playing bag file and do roslaunch) or play the bag file before do roslaunch ? both of them i already try it but still the same. if you have any idea ill be really grateful.

csl1994, 2019-05-07:
> @csl1994 hay~ i also have same problem as you did. do you solved the problem ? @johnjin404 i already your suggest try to wait for bag file playing until the end but nothing change. is it need to do [...]

closed 2019-05-07 by csl1994

csl1994, 2019-05-07:
@maxwu8416 I think the parameter("max_time_min") means that the bag file should be played for 200 minutes, so I changed it to 1 minute and it will be finished soon, but it will remind me that "the time is too short".finally i run imu for 40 m, and set the  parameter as follow
<param name="max_time_min" type="int" value= "30"/>

maxwu8416, 2019-05-07:
Okay~ seems worth to try ! Thx for ur reply ! I’ll try it later and update the info

foemre, 2022-08-21:
It seems that the program doesn't give useful feedback about whether it receives imu info. But it does. So if you cannot be sure, just change `max_time_min` in your launch file to 1 min or so, and see the results. But for accurate results you have to at least run it for about 2 hours.

tulingcheng86, 2024-01-26:
> @maxwu8416我认为参数(“max_time_min”)的意思是bag文件要播放xxx分钟，所以我把它改成1分钟，很快就播放完了，但它会提醒我“时间太短”。最后我运行imu 40 m，参数设置如下 `<param name="max_time_min" type="int" value= "30"/>`

THINKS! IT HELPFUL!

tulingcheng86, 2024-01-26:
> @maxwu8416我认为参数(“max_time_min”)的意思是bag文件要播放xxx分钟，所以我把它改成1分钟，很快就播放完了，但它会提醒我“时间太短”。最后我运行imu 40 m，参数设置如下 `<param name="max_time_min" type="int" value= "30"/>`

有用！

wells-sde, 2024-11-26:
> > @maxwu8416我认为参数(“max_time_min”)的意思是bag文件要播放xxx分钟，所以我把它改成1分钟，很快就播放完了，但它会提醒我“时间太短”。最后我运行imu 40 m，参数设置如下 `<param name="max_time_min" type="int" value= "30"/>`
> 
> THINKS! IT HELPFUL!

你帧的需要等30分钟吗，还是只要imu bag 播放完了就可以计算参数了？