gaowenliang/imu_utils #19: imu measurement rate and rosbag play rate

mmp52, 2020-01-09:
Hello @gaowenliang ,

My imu works on 50 hz, if I record the values on my rosbag I can see that recording happens not exactly at 50hz, so in order to work on this algorithm shoud I play rosbag with rosbag play -r mybagfile.bag anyway, or should I directly play it without indicating the frequency?
Also, does this algorithm work on only 200 hz or some other frequencies as well, if so should I give the frequency of my IMU as an input to the system somewhere?

Thanks,
Metin

tianxinyu0509, 2020-01-10:
@mmp52 Regardless of the frequency of the imu, ros sends an imu message with a timestamp.The code is processed based on this timestamp. Directly play it without indicating the frequency.

mmp52, 2020-01-10:
@tianxinyu0509 Thank you very much!

closed 2020-01-10 by mmp52