gaowenliang/imu_utils #22: How to transfer my IMU data to a rosbag？

Nronaldo, 2020-05-15:
Hello. I am a newbie to ROS. I appreciate your nice work. 
But I wonder how to transfer my IMU data to a rosbag?  I use my Android phone to collect IMU measurements and I  want to analyze them.

AfterJourney00, 2020-08-12:
rosbag record [topic]

sysuhuang, 2020-08-27:
> Hello. I am a newbie to ROS. I appreciate your nice work.
> But I wonder how to transfer my IMU data to a rosbag? I use my Android phone to collect IMU measurements and I want to analyze them.

Bro, have you solved thie problem?

Nronaldo, 2020-08-27:
> > Hello. I am a newbie to ROS. I appreciate your nice work.
> > But I wonder how to transfer my IMU data to a rosbag? I use my Android phone to collect IMU measurements and I want to analyze them.
> 
> Bro, have you solved thie problem?

Hello，you can use the bagcreater of kalibr

sysuhuang, 2020-08-27:
> > > Hello. I am a newbie to ROS. I appreciate your nice work.
> > > But I wonder how to transfer my IMU data to a rosbag? I use my Android phone to collect IMU measurements and I want to analyze them.
> > 
> > 
> > Bro, have you solved thie problem?
> 
> Hello，you can use the bagcreater of kalibr

Is it convenient for you to discuss more details， as I am new in IMU process, by email (or QQ/wechat as you also seems to come from China).

Nronaldo, 2020-08-27:
> Is it convenient for you to discuss more details， as I am new in IMU process, by email (or QQ/wechat as you also seems to come from China).

kalibr 的wiki上有bagcreater 的用法，这个是kalibr的工具之一，既可以把IMU和图像同时转成bag，也可以处理只有IMU的情况，当然你得把你的IMU格式整理成bagcreater的格式。

sysuhuang, 2020-09-05:
> kalibr 的wiki上有bagcreater 的用法，这个是kalibr的工具之一，既可以把IMU和图像同时转成bag，也可以处理只有IMU的情况，当然你得把你的IMU格式整理成bagcreater的格式。

老哥有用过IMU_TK这个工具么 kalibr我感觉算的太差了