gaowenliang/imu_utils #30: 单位错了吧？

yandld, 2020-11-20:
acc和 gyr 的 零偏稳定性 单位错了吧

voyage19941119, 2021-03-09:
> acc和 gyr 的 零偏稳定性 单位错了吧

acc gyro bias instability 在这篇git里单位是 (rad /s ) * sqrt(Hz) , 和 （m/s2）*sqrt(Hz)， 等价为 rad/s2/sqrt(Hz), m/s3/sqrt(Hz)
与 [kalib](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model) 里的单位一致， 也和 Indirect Kalman Filter For 3d Attitude Estimation 里推导的 bias 对时间导数的标准差单位一致。
我理解，这是bias 对时间导数的标准差，而不是bias标准差。如果说的不对请 @gaowenliang 指正

voyage19941119, 2021-03-09:
而查看Xsense手册，上面写的 In-run bias stability 单位是 deg/h, m/s2. 我理解, 这是离散化后的单位， 而这篇git里的是连续时间的单位。 参考Indirect Kalman Filter公式 (129)(130)
[image]

zhaozhongch, 2021-07-16:
我差点也看错了。公式是正确的，谢谢解释