gaowenliang/imu_utils #31: Error in evaluating the ResidualBlock.

chunibyo-wly, 2020-12-09:
First of all, thanks for your work. However I have met some problem.

ros bag which record imu topic is [imu.bag](https://drive.google.com/file/d/1xVBIqk1f_WOZPTQW5X3ebYnwSGI7cPfp/view?usp=sharing)

the full log is here [imu_utils log file](https://drive.google.com/file/d/17_aWd2aIEv3cbi3e1UplgjU4go06oc-8/view?usp=sharing)

launch file is [jy901.launch](https://drive.google.com/file/d/13M2uFex0169R3eeYKj38_2aEh12whmTn/view?usp=sharing)

and the answer is [imu.yaml](https://drive.google.com/file/d/1Td2_fyY7AxTqoNyL-wf4GNinuN-UtZQW/view?usp=sharing)

```
Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all    
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation. 

Residual Block size: 1 parameter blocks x 1 residuals

For each parameter block, the value of the parameters are printed in the first column   
and the value of the jacobian under the corresponding residual. If a ParameterBlock was 
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry 
of the Jacobian/residual array was requested but was not written to by user code, it is 
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating 
to Inf or NaN is also an error.  

Residuals:             -nan 

Parameter Block 0, size: 5

           0 |         -nan 
           0 |         -nan 
           0 |         -nan 
           0 |         -nan 
           0 |         -nan 
```

thanks for your help

Lmixq, 2021-01-19:
Hi, I meet the same problem. Do you solve it? Thank you!

chunibyo-wly, 2021-01-19:
> Hi, I meet the same problem. Do you solve it? Thank you!

I have checked my dataset, and I found that when IMU(jy901) was stationary, data from gyro was always 0, and caused error.

Obviously, it's robot's error.

Lmixq, 2021-01-19:
> > Hi, I meet the same problem. Do you solve it? Thank you!
> 
> I have checked my dataset, and I found that when IMU(jy901) was stationary, data from gyro was always 0, and caused error.
> 
> Obviously, it's robot's error.

Thanks for your reply！

As you said, the acc and gyro is loss. And the data is from xsens MT300,  I check the output setting on MT manager. After picking the option of delta_v and delta_q, I get the output in acc and gyro.

Thanks a lots!

closed 2021-01-19 by chunibyo-wly

daibo-grad, 2021-03-29:
> > Hi, I meet the same problem. Do you solve it? Thank you!
> 
> I have checked my dataset, and I found that when IMU(jy901) was stationary, data from gyro was always 0, and caused error.
> 
> Obviously, it's robot's error.

Hello, How did you solve the problem? I recorded the rosbag and data from gryo was always 0, too. How should I solve the problem?

chunibyo-wly, 2021-03-31:
No, I still have this problem.
And I have checked jy901 driver code on robot.
gyro raw data reads from serial is 0 when static.
But euler angle data is change all the time.
```cpp
    case 0x52: {
      temp_16 = (static_cast<int16_t>(chrTemp[3])<<8)|chrTemp[2];
      stcGyro.w[0] = static_cast<double>(temp_16)/32768.0*2000.0;
      temp_16 = (static_cast<int16_t>(chrTemp[5])<<8)|chrTemp[4];
      stcGyro.w[1] = static_cast<double>(temp_16)/32768.0*2000.0;
      temp_16 = (static_cast<int16_t>(chrTemp[7])<<8)|chrTemp[6];
      stcGyro.w[2] = static_cast<double>(temp_16)/32768.0*2000.0;
      temp_16 = (static_cast<int16_t>(chrTemp[9])<<8)|chrTemp[8];
      stcAcc.T = static_cast<double>(temp_16)/100.0;

      std::cout << "Temp: " << stcGyro.T << ", " << "Gyro: " << stcGyro.w[0] << ", " << stcGyro.w[1] << ", " << stcGyro.w[2] << std::endl;
      break;
    }
```

daibo-grad, 2021-03-31:
Hello, I solved the same problem a few days ago. The device was  mynteye binocular camera, and I used the device to record the IMU raw data three times.For the first time, raw data of gyro x, gyro y and gyro z was all 0. For the second time, only raw data of gyro y was 0.In the last time, raw data of gyro x, gyro y and gyro z did not equal to 0 and changed over time. Maybe you can try to record the IMU raw data several times. In my opinion, the possible cause og the problem is listed below:
https://github.com/gaowenliang/imu_utils/issues/3
"seems that data you collected is not suitable for the Ceres Solver. Can you provide your data for debug?"

 

     case 0x52: {       temp_16 = (static_cast<int16_t&gt;(chrTemp[3])<<8)|chrTemp[2];       stcGyro.w[0] = static_cast<double&gt;(temp_16)/32768.0*2000.0;       temp_16 = (static_cast<int16_t&gt;(chrTemp[5])<<8)|chrTemp[4];       stcGyro.w[1] = static_cast<double&gt;(temp_16)/32768.0*2000.0;       temp_16 = (static_cast<int16_t&gt;(chrTemp[7])<<8)|chrTemp[6];       stcGyro.w[2] = static_cast<double&gt;(temp_16)/32768.0*2000.0;       temp_16 = (static_cast<int16_t&gt;(chrTemp[9])<<8)|chrTemp[8];       stcAcc.T = static_cast<double&gt;(temp_16)/100.0;       std::cout << "Temp: " << stcGyro.T << ", " << "Gyro: " << stcGyro.w[0] << ", " << stcGyro.w[1] << ", " << stcGyro.w[2] << std::endl;       break;     }

wz8530, 2023-07-28:
@chunibyo-wly Have you solved it? My imu model is JY901B.