gaowenliang/imu_utils #32: CV errors

Rotoslider, 2021-02-17:
I am running Ubuntu 18.04 on a Jetson AGX with OpenCV4
I get the following error when trying to build cv_utils:
```
/home/lbackpack/catkin_ws/src/code_utils/src/mat_io_test.cpp: In function ‘int main()’:
/home/lbackpack/catkin_ws/src/code_utils/src/mat_io_test.cpp:33:47: error: ‘CV_LOAD_IMAGE_UNCHANGED’ was not declared in this scope
     Mat img1 = imread( "/home/gao/IMG_1.png", CV_LOAD_IMAGE_UNCHANGED );
                                               ^~~~~~~~~~~~~~~~~~~~~~~
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp: In function ‘void test1()’:
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp:85:47: error: ‘CV_LOAD_GRAYSCALE’ was not declared in this scope
     Mat img1 = imread( "/home/gao/IMG_1.png", CV_LOAD_GRAYSCALE );
                                               ^~~~~~~~~~~~~~~~~
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp:85:47: note: suggested alternative: ‘CV_HAL_DFT_SCALE’
     Mat img1 = imread( "/home/gao/IMG_1.png", CV_LOAD_GRAYSCALE );
                                               ^~~~~~~~~~~~~~~~~
                                               CV_HAL_DFT_SCALE
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp:95:35: error: ‘CV_MINMAX’ was not declared in this scope
     normalize( img, img2, 0, 255, CV_MINMAX );
                                   ^~~~~~~~~
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp:95:35: note: suggested alternative: ‘CV_FINAL’
     normalize( img, img2, 0, 255, CV_MINMAX );
                                   ^~~~~~~~~
                                   CV_FINAL
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp: In function ‘void test2()’:
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp:108:47: error: ‘CV_LOAD_IMAGE_GRAYSCALE’ was not declared in this scope
     Mat img1 = imread( "/home/gao/IMG_1.png", CV_LOAD_IMAGE_GRAYSCALE );
                                               ^~~~~~~~~~~~~~~~~~~~~~~
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp:118:35: error: ‘CV_MINMAX’ was not declared in this scope
     normalize( img, img2, 0, 255, CV_MINMAX );
                                   ^~~~~~~~~
/home/lbackpack/catkin_ws/src/code_utils/src/sumpixel_test.cpp:118:35: note: suggested alternative: ‘CV_FINAL’
     normalize( img, img2, 0, 255, CV_MINMAX );
                                   ^~~~~~~~~
                                   CV_FINAL
code_utils/CMakeFiles/sumpixel_test.dir/build.make:81: recipe for target 'code_utils/CMakeFiles/sumpixel_test.dir/src/sumpixel_test.cpp.o' failed
make[2]: *** [code_utils/CMakeFiles/sumpixel_test.dir/src/sumpixel_test.cpp.o] Error 1
CMakeFiles/Makefile2:949: recipe for target 'code_utils/CMakeFiles/sumpixel_test.dir/all' failed
make[1]: *** [code_utils/CMakeFiles/sumpixel_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
code_utils/CMakeFiles/matIO_test.dir/build.make:81: recipe for target 'code_utils/CMakeFiles/matIO_test.dir/src/mat_io_test.cpp.o' failed
make[2]: *** [code_utils/CMakeFiles/matIO_test.dir/src/mat_io_test.cpp.o] Error 1
CMakeFiles/Makefile2:841: recipe for target 'code_utils/CMakeFiles/matIO_test.dir/all' failed
make[1]: *** [code_utils/CMakeFiles/matIO_test.dir/all] Error 2
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```
I am very new to linux and am not sure how to fix this. All my attempts so far have failed in the last two weeks.
Google search has not been helpful.

thanks in advance

Rotoslider, 2021-02-17:
I fixed the cv errors by changing all instances of CV_LOAD_IMAGE_GRAYSCALE to IMREAD_GRAYSCALE
All instances of CV_MINMAX to cv::NORM_MINMAX
You need to change this in mat_io_test.cpp and sumpixel_test.cpp
My fork has the changes to allow to run on opencv4 if anyone is interested

p-CXTN, 2022-09-05:
Also change `CV_LOAD_IMAGE_UNCHANGED` to `IMREAD_UNCHANGED`

ZhangzrJerry, 2024-11-15:
CV_MINMAX -> CV_MMX
CV_LOAD_IMAGE_UNCHANGED -> IMREAD_UNCHANGED
works