gaowenliang/imu_utils #39: I build this on Ubuntu20.04 ROS noetic, and get error...

fengyikawhi, 2022-02-15:
error: no type named ‘Parameters’ in ‘class ceres::internal::ParameterDims<false, 5>’
  315 |   using Parameters = typename ParameterDims::Parameters;
      |         ^~~~~~~~~~
/home/slam/cartographer_ws/install_isolated/include/ceres/internal/autodiff.h:348:47: error: no type named ‘Parameters’ in ‘class ceres::internal::ParameterDims<false, 5>’
  348 |   Make1stOrderPerturbations<Parameters>::Apply(parameters,
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
  349 |                                                parameters_as_jets.data());
      |                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/slam/cartographer_ws/install_isolated/include/ceres/internal/autodiff.h:357:39: error: no type named ‘Parameters’ in ‘class ceres::internal::ParameterDims<false, 5>’
  357 |   Take1stOrderParts<Parameters>::Apply(
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  358 |       num_outputs, residuals_as_jets.data(), jacobians);
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

DreamWaterFound, 2022-02-15:
Maybe you need to compile with the correct version like 2.0.0 of Ceres-solver.

fengyikawhi, 2022-02-16:
> Maybe you need to compile with the correct version like 2.0.0 of Ceres-solver.

I have installed Ceres2.0.0 when build cartographer, but it was in the workspace of cartographer, does it matter?

DreamWaterFound, 2022-02-19:
> > Maybe you need to compile with the correct version like 2.0.0 of Ceres-solver.
> 
> I have installed Ceres2.0.0 when build cartographer, but it was in the workspace of cartographer, does it matter?

Yes, in your case the compiler will only compile `imu_utils` with ceres-solver in the default path. 

There are 2 ways to solve this problem:
1. To complie and install ceres-slover in the same ROS workspace with `imu_utils`;
2. Specify the path where ceres-solver 2.0 installed in `CMakeLists.txt`(Suggested methods). You can locate `CeresConfig.cmake` using command `locate`:

```bash
$ sudo updatedb
$ locate CeresConfig.cmake
```
Find the installation path for Ceres-Solver 2.0, like `/home/robot/Libraries/Install/Ceres/2.0.0/lib/cmake/Ceres/CeresConfig.cmake`.
Then add following line in [CMakeLists.txt](https://github.com/gaowenliang/imu_utils/blob/master/CMakeLists.txt):

```cmake
...

find_package(OpenCV REQUIRED)

# HACK add here
# HACK begin
set(Ceres_DIR "/home/robot/Libraries/Install/Ceres/2.0.0/lib/cmake/Ceres/CeresConfig.cmake")
# HACK end

find_package(Ceres REQUIRED)

find_package(Eigen3  REQUIRED)
```

Be sure clean folders `build` and `devel` in this ROS worksapce and then re-build it by `catkin_make`. In this way gcc/g++ will use you specified ceres-solver.