gaowenliang/imu_utils #47: build problem

brian890815, 2024-01-08:
Hi, I got the following build error (after catkin_make):
what could be the problem?

CMake Error at /home/vccl/catkin_ws/build/ros_comm/clients/roscpp/cmake/roscpp-genmsg.cmake:79 (add_custom_target):
  add_custom_target cannot create target "roscpp_generate_messages_cpp"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/home/vccl/catkin_ws/src/imu_utils".  See documentation for policy CMP0002
  for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  ros_comm/clients/roscpp/CMakeLists.txt:41 (generate_messages)

CMake Error at /home/vccl/catkin_ws/build/ros_comm/clients/roscpp/cmake/roscpp-genmsg.cmake:136 (add_custom_target):
  add_custom_target cannot create target "roscpp_generate_messages_eus"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/home/vccl/catkin_ws/src/imu_utils".  See documentation for policy CMP0002
  for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  ros_comm/clients/roscpp/CMakeLists.txt:41 (generate_messages)

CMake Error at /home/vccl/catkin_ws/build/ros_comm/clients/roscpp/cmake/roscpp-genmsg.cmake:193 (add_custom_target):
  add_custom_target cannot create target "roscpp_generate_messages_lisp"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/home/vccl/catkin_ws/src/imu_utils".  See documentation for policy CMP0002
  for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  ros_comm/clients/roscpp/CMakeLists.txt:41 (generate_messages)

CMake Error at /home/vccl/catkin_ws/build/ros_comm/clients/roscpp/cmake/roscpp-genmsg.cmake:250 (add_custom_target):
  add_custom_target cannot create target "roscpp_generate_messages_nodejs"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/home/vccl/catkin_ws/src/imu_utils".  See documentation for policy CMP0002
  for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  ros_comm/clients/roscpp/CMakeLists.txt:41 (generate_messages)

CMake Error at /home/vccl/catkin_ws/build/ros_comm/clients/roscpp/cmake/roscpp-genmsg.cmake:307 (add_custom_target):
  add_custom_target cannot create target "roscpp_generate_messages_py"
  because another target with the same name already exists.  The existing
  target is a custom target created in source directory
  "/home/vccl/catkin_ws/src/imu_utils".  See documentation for policy CMP0002
  for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  ros_comm/clients/roscpp/CMakeLists.txt:41 (generate_messages)

-- +++ processing catkin package: 'navigation'
-- ==> add_subdirectory(navigation)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'octomap_server'
-- ==> add_subdirectory(octomap_mapping/octomap_server)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'navi_multi_goals_pub_rviz_plugin'
-- ==> add_subdirectory(rviz_navi_multi_goals_pub_plugin/navi_multi_goals_pub_rviz_plugin)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using Qt5 based on the rviz_QT_VERSION: 5.9.5
-- +++ processing catkin package: 'dwa_local_planner'
-- ==> add_subdirectory(dwa_local_planner)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'xsens_mti_driver'
-- ==> add_subdirectory(xsens_ros_mti_driver)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Configuring incomplete, errors occurred!
See also "/home/vccl/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/vccl/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

ESSEOGHENNA, 2024-11-10:
bro pls give me something more exclusive

ZhangzrJerry, 2024-11-15:
try to build like this
```bash
catkin_make -DCATKIN_WHITELIST_PACKAGES="imu_utils"
catkin_make -DCATKIN_WHITELIST_PACKAGES="code_utils" 
```